neka-nat/freecad-mcp #1: Error: spawn uv ENOENT

blindFS, 2025-03-30:
First of all, thanks for creating this amazing tool. 
But I have trouble initializing it, hope you can give me some hint of how to fix.

Here's my setup procedures.
 
1. I installed both FreeCAD and Claude with homebrew on my mac.
2. Followed the steps of addon installation, here's what I get, somewhat different from your screenshot, but I guess it's not a big issue.

[image]

3. When I pressed the `start MCP server` button, I got these from the Report View of FreeCAD (seems started successfully):

```
11:03:15  RPC Server started at localhost:9875
11:03:15  RPC Server started at localhost:9875.
```

4. Then I opened the Claude app with the exact json configuration from README, I got these errors:

[image]

```
2025-03-30T03:07:44.307Z [freecad] [info] Initializing server...
2025-03-30T03:07:44.322Z [freecad] [error] spawn uvx ENOENT {"context":"connection","stack":"Error: spawn uvx ENOENT\n    at ChildProcess._handle.onexit (node:internal/child_process:285:19)\n    at onErrorNT (node:internal/child_process:483:16)\n    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)"}
2025-03-30T03:07:44.322Z [freecad] [error] spawn uvx ENOENT {"stack":"Error: spawn uvx ENOENT\n    at ChildProcess._handle.onexit (node:internal/child_process:285:19)\n    at onErrorNT (node:internal/child_process:483:16)\n    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)"}
2025-03-30T03:07:44.324Z [freecad] [info] Server transport closed
2025-03-30T03:07:44.324Z [freecad] [info] Client transport closed
2025-03-30T03:07:44.324Z [freecad] [info] Server transport closed unexpectedly, this is likely due to the process exiting early. If you are developing this MCP server you can add output to stderr (i.e. `console.error('...')` in JavaScript, `print('...', file=sys.stderr)` in python) and it will appear in this log.
2025-03-30T03:07:44.324Z [freecad] [error] Server disconnected. For troubleshooting guidance, please visit our [debugging documentation](https://modelcontextprotocol.io/docs/tools/debugging) {"context":"connection"}
```

neka-nat, 2025-03-30:
Hi
It's likely that either the path to `uv` is not set, or `uv` is not installed.
Please check the path to `uv`.

blindFS, 2025-03-30:
> Hi It's likely that either the path to `uv` is not set, or `uv` is not installed. Please check the path to `uv`.

Installing `uv` does solve the problem, thanks very much.

closed 2025-03-30 by blindFS