neka-nat/freecad-mcp #2: FreeCAD MCP Issue

cnirbhay, 2025-04-06:
Hi @neka-nat, I followed the installation instructions. I could see the MCP Addon inside FreeCAD but nothing happens when I try triggering "Start RPC Server" option.

[image]

For setting up Claude Desktop, I tried both of the below options by editing the json file. After trying first option: nothing happens; and after trying second option: nothing happens and I see an error message when opening Claude as shown below.

[image]

[image]

{ "mcpServers": { "freecad": { "command": "uv", "args": [ "--directory", "C:\Users\home\AppData\Roaming\FreeCAD\Mod\FreeCADMCP", "run", "freecad-mcp" ] } } }

Following are my system details:

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/India (en_IN)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:

ArchTextures
Curves 0.6.53
freecad.gears 1.3.0
FreeCADMCP
Nodes 0.1.36
Reinforcement
StructureTools

cnirbhay, 2025-04-06:
@neka-nat thanks for your response. I tried like following but no success. Please review.

[image]

`{
  "mcpServers": {
    "freecad": {
      "command": "uv",
      "args": [
        "--directory",
        "G:\Softwares\freecad-mcp",
        "run",
        "freecad-mcp"
      ]
    }
  }
}`

neka-nat, 2025-04-06:
Is uv installed?

neka-nat, 2025-04-06:
Also, from the error message, it looks like there's something wrong with the way the JSON is written.

cnirbhay, 2025-04-06:
> Is uv installed?

Yes, uv is installed.

cnirbhay, 2025-04-06:
> Also, from the error message, it looks like there's something wrong with the way the JSON is written.

Please advise for correction.

neka-nat, 2025-04-07:
As stated in the error message, there seems to be an invalid escape sequence at line 7, column 13.
[image]

cnirbhay, 2025-04-07:
What would be the right json then?

neka-nat, 2025-04-07:
That's hard to tell because we don't have the actual JSON file.
It might be a good idea to ask ChatGPT or something similar.

cnirbhay, 2025-04-07:
I tried that also and it suggested to use double backslashes. This time I didn't get the error message inside Claude but nothing happens when I trigger "Start RPC Server" option from FreeCAD.

cnirbhay, 2025-04-07:
It provided some steps like following which I performed but unfortunately no success.

[image]

[image]

[image]

neka-nat, 2025-04-07:
What kind of response is Claude providing to your question?

neka-nat, 2025-04-07:
The JSON settings aren't good. Please do the following.

```json
{
  "mcpServers": {
    "freecad": {
      "command": "uvx",
      "args": [
        "freecad-mcp"
      ]
    }
  }
}
```

cnirbhay, 2025-04-08:
It worked now after I started my PC today. The main issue I realised was that I was editing my Claude config file after closing Claude but Claude was never actually fully closed, so I "Quit" it from its hidden icon in the taskbar.

Just one quick query: I tried generating geometries from Claude but it doesn't stop on its own and continue to create multiple times. I had to manually stop the generation or tweak the prompt.

[image]

neka-nat, 2025-04-08:
I'm glad to hear that.
If you liked this software, I’d really appreciate it if you could give it a star.
You might need to tell the prompt to "create just one."
They tend to be thoughtful and generate various model patterns.

cnirbhay, 2025-04-08:
Yup sure, done!
Thanks for the help, and I'm closing this issue.

closed 2025-04-08 by cnirbhay