neka-nat/freecad-mcp #6: PermissionError

zaizai0935, 2025-04-23:
When the MCP performs a task, it cannot read the PNG files in %temp%, for example:
Failed to get objects: <Fault 1: "<class 'PermissionError'>:[WinError 32] The file is being used by another process and cannot be accessed: 'C:\\Users\\username\\AppData\\Local\\Temp\\tmpgch23w0m.png'">

neka-nat, 2025-04-23:
### Root Cause  
In `FreeCADRPC.get_active_screenshot()` you create the temp file like this:

```python
temp_file = tempfile.NamedTemporaryFile(suffix=".png", delete=False)
rpc_request_queue.put(lambda: self._save_active_screenshot(temp_file.name, view_name))
```

On **Windows, `NamedTemporaryFile` keeps an exclusive lock until it is closed**, so when the
FreeCAD GUI tries to write the PNG to the same path it raises

```
PermissionError: [WinError 32] The file is being used by another process
```

### Fix (pick one)

1. **Close the file immediately**

```python
temp_file = tempfile.NamedTemporaryFile(suffix=".png", delete=False)
path = temp_file.name
temp_file.close()                    # ★ release the lock early
rpc_request_queue.put(lambda: self._save_active_screenshot(path, view_name))
...
```

2. **Use `mkstemp` to get a path only**

```python
fd, path = tempfile.mkstemp(suffix=".png")
os.close(fd)                         # ★ close the descriptor
rpc_request_queue.put(lambda: self._save_active_screenshot(path, view_name))
...
```

Both variants **release the handle before FreeCAD writes the image**, eliminating the
`WinError 32`.

#### Further best-practice tweaks
| Step                         | Why it helps                               |
|------------------------------|--------------------------------------------|
| `with open(path, "rb")`      | guarantees the file is closed after read   |
| `finally: os.remove(path)`   | ensures cleanup even if an exception occurs|
| `.saveImage(path, dpi)`      | fixing DPI (e.g. 100) yields stable size   |

With these changes the MCP should be able to read and delete screenshots in `%TEMP%`
without any locking errors.

neka-nat, 2025-04-23:
Thanks for the report!
I’ve fixed the image save handling, so please try the latest version of the main branch.

zaizai0935, 2025-04-23:
Great job! Thank you very much!

closed 2025-04-23 by neka-nat