neka-nat/freecad-mcp #23: May not work under mcp version < 1.12.4

dddai, 2025-08-14:
I want to report that tool may not be working under mcp version > 1.12.2 in windows claude desktop.

It seems that error in below URL occurring.
https://github.com/modelcontextprotocol/python-sdk/issues/1234

In my environment, this tool is not working.
looking into Claude logfile (/logs/mcp-server-freecad.log) , I found that python throwing error.

##log clip start##
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\AppData\Local\uv\cache\archive-v0\TA5oPOuSPEgQPpYmyJriX\Scripts\freecad-mcp.exe\__main__.py", line 4, in <module>
  File "C:\Users\AppData\Local\uv\cache\archive-v0\TA5oPOuSPEgQPpYmyJriX\Lib\site-packages\freecad_mcp\server.py", line 114, in <module>
    mcp = FastMCP(
          ^^^^^^^^
TypeError: FastMCP.__init__() got an unexpected keyword argument 'description'

##log clip end##

in corresponding code at specified line, I found description parameter given as below.

####code clip start##
mcp = FastMCP(
    "FreeCADMCP",
    description="FreeCAD integration through the Model Context Protocol",
    lifespan=server_lifespan,
)
####code clip end##

I searched in google and found below URL.
https://github.com/modelcontextprotocol/python-sdk/issues/1234

I'm not sure where this bug came from (it maybe claude desktop problem...)
but I thought it is better to report this issue at least. 
(Corresponding code is in cached directory and not able to trace which code is copying or creating this chunk of code.)

Depends on root cause...but if it is claude problem, it maybe better to specify  mcp versions.

Thanks

neka-nat, 2025-08-17:
Thank you for reporting a bug.
I’ve fixed it. Please try the latest main branch.

dddai, 2025-08-19:
Thanks for bug fix.
I tried the fix code and confirmed that it actually works.

Thank you.